brumster/EDTracker2 #2: No Gyro defined

patrickpoirier51, 2016-03-28:
Hello,
I get this error when trying to compile EDTracker2_9250 
arduino-1.6.6/libraries/InvensenseMotionDriver/inv_mpu.c:122:2: error: #error Which gyro are you using? Define MPUxxxx in your compiler options.
 #error  Which gyro are you using? Define MPUxxxx in your compiler options.

Note:
From your last mod, the define MPU9250 has been removed?

Regards

patrickpoirier51, 2016-03-28:
Made some progress:  added  #define MPU9250 in /avr/cores/arduino/Arduino.h
Now, I get this error: EDTracker2_9250:173: error: 'TrackState_t' does not name a type
 TrackState_t joySt;

brumster, 2016-03-29:
It's defined in the hardware files, so controlled by the device you choose in your Arduino IDE. It's not needed in the main sketch. Sounds to me like you have old hardware files, or have the wrong device type selected?

patrickpoirier51, 2016-03-29:
It is working now... Just had to read carefully where to store the hardware files :-( 
I suggest that you add this to the main README:
Place the "edtracker" folder into your <Arduino Install>/hardware/ folder. 
Thanks for your support and this great product

brumster, 2016-03-30:
No problems; will accept as a doco defect and address soon... ;)

closed 2016-05-22 by patrickpoirier51