brumster/EDTracker2 #3: MPU 9250 calculates Gyro Bias forever

shad00m, 2017-05-13:
I've went to DIY way and got the arduino with an MPU-9250.
Flashing 9250 4.0.5 leaves me at "Calculating Gyro Bias" forever, it wont stop.
I can flash 6050, Calib, and even the 9250_old 4.0.3.
But in the _old the Magnetometer and the temperature doesn't seem to work.
In the 6050 the temperature does work though.

Do I have a fake MPU-9250? Is it a firmware error?

shad00m, 2017-05-13:
I also had this error when starting it the first time on 3 different machines:
https://hastebin.com/xigeqopuve.sql

brumster, 2017-05-15:
Unlikely to be a firmware issue as it's been stable for so long.

I've seen your post on the ED forums, with the "Mag oops [7]" message. 7 is the error return code from the Invensense Motion Driver libraries (see inv_mpu.c). It seems to suggest there is either a data overrun or the data ready bit is not being set - probably the latter at a guess.

This suggests a hardware issue or an issue with the i2c timing, but if other non-mag sketches are showing sensible behaviour then it does look like a faulty mag on the MEMS IC.

Will leave this open for you to feedback for a little longer, but this doesn't appear as a software issue... I cannot recreate.

shad00m, 2017-05-15:
Thank you for your help brumster. I ordered a new MPU from a different, non-chinese seller. It should come in 1-2 days. I'll update this issue when it arrives.

AlecVoid, 2017-05-17:
Hi, 
I can confirm this bug. 

I tested the EDTracker2_9250.ino with a functional EDTracker (it works perfectly if you are flashing with EDTrackerUI4 v4.0.4.0) and it is exactly the same problem: Auto gyro bias never stops and it spams "Mag oops [7]" to the serial console if you uncomment the DEBUG line.

I tried Arduino IDE 1.8.1, 1.8.2 and 1.6.7 .. everytime the same problem.

It is definitly a problem with the software - not hardware (because the hardware works fine with the UI .. mentioned before).

Brumster, I hope you will find the time to debug this code. Keep on the good work!

Best regards

shad00m, 2017-05-18:
New MPU arrived. Looks exactly the same even the same wrongly rotated "GND" label.
and it works flawless... magnetometer, temperature, auto gyro bias....

AlecVoid, 2017-05-18:
@shad00m 
Could you please check if the problem shows again if you flash the edtracker using the Arduino IDE?

shad00m, 2017-05-18:
@AlecVoid 
Nothing appears in the Serial Console now.

AlecVoid, 2017-05-18:
Did you flash it with the Arduino IDE or with the EDTracker UI? I want to confirm the problem that I found with the current EDTracker source.

Please flash with the Arduino IDE and after that please check if all works normal in the EDTracker UI.

shad00m, 2017-05-18:
Opened Arduino 1.6.7 - loaded the 9250 sketch - clicked upload - it finishes with a couple of notes and warnings (https://hastebin.com/cowoponiho.tex)
After that I open EdTrackerUI404 and everything seems to be working fine.
This is the correct way to flash through arduino right? I'm not familiar with the Arduino IDE

AlecVoid, 2017-05-18:
@shad00m 
You did a perfect job. That was exactly what I wanted to be tested. 

This bug seems to be a kind of poltergeist .. you had it and now I have it. ;)

brumster, 2017-07-14:
Just tidying up here - it seems from the above this was a hardware issue, so can it be closed?

AlecVoid, 2017-07-14:
Yes, looks like it. Thanks again. :)

brumster, 2017-07-14:
Many thanks for confirmation. Closed as hardware issue.

sir-maniac, 2017-09-21:
I'm getting this issue, and it seems repeatable.  

If I compile and upload the EDTracker2_9250 v4.0.5, using the 9250 board type(both cloned from the repos), I get the "Mag oops [7]".  When I flash it from the EDTrackerUI, it works fine.

I tried commenting out code the lines that check for this error bit in inv_mpu.c, and it allows one to proceed, but it gets no readings when attempting to calibrate the magnetometer.

I'm using Arduino IDE v1.8.4.  

I'll try an earlier version of Arduino and see if it has the same problem.

brumster, 2017-09-23:
Yes, our suspicions are that some MPU boards or actual ICs have faulty or marginal magnetometers. It seems they report sensor values fine apart from the mag and sometimes the temperature. There is a slim chance it's a construction issue but I'm hard pushed to explain how i2c lines communication fine except on specific register values; that doesn't make much sense! As you've discovered, ignoring the "Mag Oops" error in the code (something passed up from the Invensense MPU libraries) would just move the issue elsewhere, as you would have no idea that the mag wasn't reporting anything until you tried to calibrate it. I've asked Rob if he can work with me to make the UI better at reporting this hardware fault to the end user so they know what the problem is, but he's very busy at the moment.

sir-maniac, 2017-09-23:
I'm not sure this explains why the binary version always works, and the self-compiled sketch of the same version always fails.  

Even if the chip I have only works marginally, it consistently fails when I compile it myself. That would suggest something different, if not in the code, then in the output of the compiler.

AlecVoid, 2017-09-23:
I had exactly the same issue some month back. Using the EDTrackerUI the sketch worked. Self compiled not so much.
@Brumster could you check if the binaries of both sketches are identical? If yes, my next suspicion would be the flashing method... maybe there is a difference regarding "empty" space (one way would clean that space - the other doesn't). Just an idea. :)

brumster, 2017-09-23:
Ah ok, apologies, I misunderstood. I'll do some investigation. Just for me to try to recreate, what version of the IDE are you guys using?

AlecVoid, 2017-09-23:
I tried different IDEs (as I wrote on the 17th of May):

`I tried Arduino IDE 1.8.1, 1.8.2 and 1.6.7 .. everytime the same problem.`

sir-maniac, 2017-09-23:
I'm using Arduino 1.8.4, I haven't tried other versions yet.

I've spent some free time attempting to merge the Arduino 1.8.4 core, and the motion driver to 6.1.  I have it compiling and running, and have it reporting numbers via serial, but the reporting via serial appears slow, and I haven't got the joystick reporting working yet.

I'll play around with timings and report back any findings.

sir-maniac, 2017-09-24:
With the attached patch, I'm able to compile, upload and successfully use EdTracker2_9250 v4.05.  Compiling using arduino IDE 1.8.4.

I was able to port the customizations in motion driver 5.1 to 6.12 using an original copies of the drivers for reference.  

I don't quite know where the problem was, but I suspect, from the changes in the code, some initialization values weren't quite right for newer MPU9250s.  Of course, I could be wrong.

[motion_driver_update_6.12.zip](https://github.com/brumster/EDTracker2/files/1327032/motion_driver_update_6.12.zip)

brumster, 2017-09-25:
Ok, I've recreated here - it's definitely something that the Arduino IDE has introduced post-1.6.7
I can build on 1.6.4 and 1.6.7 fine, but the minute I move to 1.8.4 with no modifications at all, I get the Mag Oops.

Now we looked to move to the later motion driver back in the early days and we found it's behaviour a bit different, which is why we specifically kept to the older 5.1 version. Let me spend some time merging your 6.12 changes in to a local copy, I just want to be absolutely 100% sure it plays the same on all versions (6050, 9150 and 9250) and then I'll adopt it in ;)

sir-maniac, 2017-09-26:
Yeah, I probably didn't get everything right anyway :)  As the compiler seems to be involved, I wonder if it's something involving timing between the chip and the processor.  If so, it may be a matter of adjusting a few numbers in the initialization.

It might be worth patching only those and see if it gets the old driver up and running.

I can understand not wanting to mess with stable code when it's worked so well.  I'll try to track that down on my end if it helps you.

brumster, 2017-09-26:
Timings would be my prime suspect too. We definitely have mixed reports of people building them today and hitting this issue; my guess would be the ICs are marginal in some area, and some examples need slightly more relaxed timings, something that the newer IDE (and therefore compiler) is more aggressive in it's approach. I'll slowly merge in 6.12 over the coming days, take it step by step, and see if I can find the bit that breaks it 😉.

Yury-MonZon, 2018-01-24:
Got the same bug. Right after fresh flashing from GUI (or from Java Updater tool) it works perfectly. Gyro bias calculation takes about 20 seconds. After few unplugging and plugging back (thus resetting the device) gyro bias calculation takes more than 3 minutes or infinity.

Yury-MonZon, 2018-01-25:
@brumster, any updates on moving to 6.12 driver?

brumster, 2018-01-25:
I had a session of moving to the 6.12 driver; it is quite a laborious process as most of the libraries were modified to reduce compile size. Unfortunately after several hours of merging changes, it compiled but did nothing - the MPU failed to initialise. So I need to re-attempt it in smaller steps and find out which bit is breaking. I have records, it's just taking time, that's all. For now what you can do is discard the MPU return value check by removing the if/then/else construct at line 292

`if (magSampled == 0)
...
#ifdef DEBUG
    } else {
      // TODO remove
      Serial.print("Mag oops [");
      Serial.print(magSampled);
      Serial.println("]");
#endif
}`

This will essentially ignore the mag read error that's happening on certain MPU chips... BUT, I suspect you might find the device is riddled with drift *if* the mag truly isn't responding all the time (essentially the mag will be disabled).

I'll revisit the code merge when I get some time and see if I can crack it...

Yury-MonZon, 2018-01-25:
Any ETA on this? I'm willing to help if you don't mind. I have no experience in working with 9250, but I have some experience with 6050 + baro (rc plane autopilot).

I can't get access to 9250 docs and driver. Can you share them?

Please, check my comment above - this is really weird, if I flash with the binary the tracker works until power cycle(reboot). What if I make the software reboot of MCU? There must be something with init delays. IMHO it looks like it.

brumster, 2018-01-27:
9250 datasheet and register maps are on the Invense website and can generally be googled OK. The driver requires membership of Invensense's developer forum, but the copies I have on file are as follows
http://www.brumster.com/downloads/motion_driver_5.1.zip
http://www.brumster.com/downloads/motion_driver_5.1.2.zip
http://www.brumster.com/downloads/motion_driver_6.12.zip

I have for a long time suspected there is something 'different' with the MPU9250 ICs coming out of invensense that makes certain ones fussy about timings. This reported "bug" makes very little sense because it doesn't affect everyone. I suspect QA issues because on our own testing we saw a large portion of 9250 chips with noisy magnetometers straight out of the factory. This included a development board supplied by Invensense themselves - two identical boards, one showed the issue, the other did not. So, mentioned in an earlier post, I have suspected timings also.

Unfortunately I don't have the exact source of the binaries on the site. They were maintained by someone else, who didn't keep his repository in line, and while I'm told I have the latest source it doesn't compile identical HEXs - however I doubt any differences are major, to be honest. You have tried on the older Arduino IDEs, right? Still doesn't work for you? You have to go 1.6.7 or earlier...

I'm going to have another go at merging in the changes this afternoon so leave it with me, but by all means have a fiddle yourself with the above libraries and let me know how you get on. The experiences I have seen over the last 12 months, coupled with the fact that the code has been stable for years, honestly makes me think Invensense are doing something weird and wonderful on the hardware, OR people are falling foul of counterfeit chips (although not sure that really exists as a problem on the MPU-9250?)... but let me work on it and see how we get on this weekend ;) !

Yury-MonZon, 2018-01-27:
Great news! Thanks for the files. I'll give it a try.

Yury-MonZon, 2018-01-27:
I've tried Arduino 1.6.5 and this time it works for me the same way as with the binary. First time it works and it stops after reset. Not a big thing, but at least it works =)

Could you point me to the reset mpu and reset mag functions that we currently have in our motion driver?

brumster, 2018-01-27:
Sure - our modified motion driver files are buried away in the "packaged up" Arduino hardware folder, in the other repo [EDTracker2_ArduinoHardware](https://github.com/brumster/EDTracker2_ArduinoHardware/tree/master/edtracker/avr/libraries/InvensenseMotionDriver). It's in inv_mpu.c

Let me know if you have any joy modifying the timeouts/etc.

brumster, 2018-01-27:
Right then!
I've merged in the changes mentioned above by @sir-maniac and built and deployed with IDE 1.8.5 to a set of test devices here, for all variants (6050, 9150 and 9250). They all seem to work fine. So I propose the following...

1. I will shortly push up the changes to the repos and deploy to the UI update mechanism as a DEV build only
2. I'll let it stew for a while, if the guys following this bug and test and verify with some feedback, that would be greatly appreciated
3. I'll then formally compile and put them onto the update site so the GUI can flash them as test firmwares, in the TEST mode.
4. Following another short period of time allowed for testing, I'll promote them in the UI as "production" firmwares that all users will see, not just DEV mode.

Yury-MonZon, 2018-01-28:
Awesome! Same for me though - works only right after the flashing, infinite bias error after the reset. The only way to make it work again is to reflash =(

I've tried adding force_reset in the setup routine. Still, same stuff.

Yury-MonZon, 2018-01-28:
After some tests, I've got a feeling that there must be something wrong with the atmega bootloader or the CDC driver. I've used simple sketch which dumps lots and lots of ascii data on the screen and then prints the time it took. Also, blinks the led for visual reference. So I have absolutely no mpu9250 stuff in there. And it also shows the same behaviour - works fine until reboot, after reboot works slower. And you can literally see the led blinking slower...

Here is the sketch: https://pastebin.com/j60EFR6n

Sorry for the trouble, Brumster.

brumster, 2018-01-28:
Understood. Well, for me, it seems a good move forward anyway in respect of compiling on the 1.8.5 IDE so I'll leave it out there for feedback from others. On a side note, it's annoying how the image size has increased over the years with the Arduino IDE bloating things out!

Yury-MonZon, 2018-01-29:
The fix for my problem is found - wrong bootloader was flashed in the Arduino Pro Micro chip. Had to reflash it using AVR ISP mk2 and Arduino IDE. So if you got something like and your board is not genuine SparkFun Pro Micro - then reflash the bootloader. As I found some of the chips from aliexpre.. and such are prone to this. Brumster, I think it would be great if you mention it somewhere, would be useful for new DIY builders.

rentaspoon, 2018-03-04:
I seem to be having these problems now, I have a new MPU 9250 from a different supplier, I'm also gonna use my current one on multiWii (drone) to see if it's your software or not, will update ASAP

brumster, 2018-03-07:
Final call from those contributors, otherwise will close as resolved...

Yury-MonZon, 2018-03-08:
Tried on 2 other DIY trackers - works like a charm. Thanks!

brumster, 2018-06-19:
Feedback has been good. Closing as resolved; thanks to all those involved.

closed 2018-06-19 by brumster

Eduardop0520, 2023-05-24:
Hello. I know it has been 5 years since this thread was closed, but I seem to have the same problem. I used Arduino 1.8.19 IDE to flash the board and the MPU9250. It gets detected by windows as a joystick correctly, but it gets stuck at calibrating the gyro in the EDTracker GUI. Using the DEBUG mode in the IDE I get the Mag Oops [7] error too and I don't see the Dev build that you guys talk about above. Is there anything else I can try? 

Thank you and sorry for bringing the thread back from the dead

brumster, 2023-05-24:
No worries. I'd suggest you try the 1.8.5 IDE first as that's what the build is "known good" on. Otherwise I'd look to i2c noise issues as the root cause, or possibly a faulty MPU, although the latter would seem unlikely/unfortunate!

Eduardop0520, 2023-05-25:
I tried using the 1.8.5 IDE but still got the same error code. I commented the line "if (magSampled == 0)" and the axes got detected in the EDTracker GUI, but as expected the magnetometer wasn't detected. I could use it like that no problem, but the x axis returns back to center automatically, so when i look to the right, the camera initially complies, but then comes back to center.

brumster, 2023-05-25:
Yeah, without the mag feeding into the equations, it won't work properly, I'm afraid. Sorry, no idea then, maybe try some other sketches to confirm the MPU is working as intended and if so, point me to them and I'll try and work out what they're doing differently?

Eduardop0520, 2023-05-26:
I don't have a lot of knowledge in coding and arduino, so what I ended up doing was reflashing my tracker as an MPU6050 and deal with the drift through the edtacker GUI. Now it works as intended (without the magnetometer unfortunately).