brumster/EDTracker2 #5: Will the 9255 work with the 9250 sketch?

ronyshtamler, 2017-07-31:


martinrogers, 2017-08-03:
Echo that question for the MPU-92/65 board which I received from HobbyComponents this week.  I have flashed with various versions and the only way this works is with the older non-magnetometer build.

brumster, 2017-08-04:
The firmware is untested with the MPU-9255 IC from Invensense. While the datasheets suggest the 9255 should be register-compatible with the 9250, I cannot confirm for definite. If someone can confirm the chip markings on the physical IC are a 9255, and that the firmware works (or doesn't), that would be useful information. Be aware that *breakout boards* marked MPU-92/65 can be misleading, these are often generic boards made in China and the printed name on the board may not be reflective of the actual IC soldered on to the board!

MPU-9255 marking on top of IC is "M925"
MPU-9250 marking on top of IC is "MP92"

martinrogers, 2017-08-04:
HobbyComponents are sure it's a 9250 IC but I'll have a look tonight to confirm what's actually printed on it, thanks.

brumster, 2018-06-19:
I tried to purchase a 9255 from China; it was advertised as such. However when it arrived it was in fact a 9250. I am finding it impossible to actually buy a 9255 development board to verify/test I'm afraid :(. I'll leave this is an open question until someone can ever actually verify it but to my knowledge, no-one is selling MPU-9255 development boards (not TRUE 9255's)...

brumster, 2026-01-03:
Closing as no progress from the community and device availability makes the approach unlikely

closed 2026-01-03 by brumster