brumster/EDTracker2 #7: GUI serial port crash

Yury-MonZon, 2018-01-24:
I couldn't find github page for the GUI sources, so I'll try here.

Whenever I try to flash the device from the GUI(running as admin on win10 with .Net 4.5 included in OS) I get the exception:
`See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.UnauthorizedAccessException: Access to the port 'COM5' is denied.
   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   at System.IO.Ports.SerialPort.Open()
   at EDTrackerUI3.flashDialog.touchPort(String port, Int32 baud)
   at EDTrackerUI3.flashDialog.ProgressForm_Shown(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnShown(EventArgs e)
   at System.Windows.Forms.Form.CallShownEvent()
   at System.Windows.Forms.Control.InvokeMarshaledCallbackDo(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2600.0 built by: NET471REL1LAST
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
EDTrackerUI4
    Assembly Version: 4.0.4.0
    Win32 Version: 4.0.4.0
    CodeBase: file:///D:/MonZon/Downloads/EDTracker2-GUI-V4.0.4/EDTrackerUI4.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
PresentationCore
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2600.0 built by: NET471REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_32/PresentationCore/v4.0_4.0.0.0__31bf3856ad364e35/PresentationCore.dll
----------------------------------------
WindowsBase
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2600.0 built by: NET471REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/WindowsBase/v4.0_4.0.0.0__31bf3856ad364e35/WindowsBase.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2600.0 built by: NET471REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
WindowsFormsIntegration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/WindowsFormsIntegration/v4.0_4.0.0.0__31bf3856ad364e35/WindowsFormsIntegration.dll
----------------------------------------
PresentationFramework
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2600.0
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/PresentationFramework/v4.0_4.0.0.0__31bf3856ad364e35/PresentationFramework.dll
----------------------------------------
System.Xaml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2600.0 built by: NET471REL1LAST
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xaml/v4.0_4.0.0.0__b77a5c561934e089/System.Xaml.dll
----------------------------------------
MathNet.Numerics
    Assembly Version: 3.2.3.0
    Win32 Version: 3.2.3.0
    CodeBase: file:///D:/MonZon/Downloads/EDTracker2-GUI-V4.0.4/MathNet.Numerics.DLL
----------------------------------------
System.Numerics
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Numerics/v4.0_4.0.0.0__b77a5c561934e089/System.Numerics.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2612.0 built by: NET471REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
PresentationFramework.Aero2
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/PresentationFramework.Aero2/v4.0_4.0.0.0__31bf3856ad364e35/PresentationFramework.Aero2.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
GlobalHotkeys
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///D:/MonZon/Downloads/EDTracker2-GUI-V4.0.4/GlobalHotkeys.DLL
----------------------------------------
Microsoft.GeneratedCode
    Assembly Version: 1.0.0.0
    Win32 Version: 4.7.2612.0 built by: NET471REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
ISymWrapper
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2556.0 built by: NET471REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_32/ISymWrapper/v4.0_4.0.0.0__b03f5f7f11d50a3a/ISymWrapper.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
`

Any hints? Basically, I'm stuck.

brumster, 2018-01-24:
Hi chap,
The UI isn't open source and I don't look after it I'm afraid. Can you contact us via the website (support.edtracker.co.uk), raise a ticket and we can pass this on to Rob who can investigate and advise.

My first suggestion you've covered (running as Admin). Assuming everything .NET-wise is patched up to date, I wonder if there are settings in Windows 10 to "lock down" a COM port from access. Either that or an issue with the Arduino drivers.

Either way, this (Git) isn't the best place to address it. I'll let you respond before closing.

Yury-MonZon, 2018-01-24:
Roger that.
Thanks.

closed 2018-01-24 by Yury-MonZon