brumster/EDTracker2 #8: Not recognizing after flashing

PeterSpam, 2018-06-18:
I've did it on my own way (diy) by using a chinese pro micro clone, the MPU9250 and the V3.0 PCB. Now I'm at the point where I have to flash the pro micro which doesn't work because every time after a succesfull run windows isn't recognizing my pro micro anymore and I have to reflash it with an empty sketch to "repair" it. I've tried to upload it manually via the IDE too but with the same effect...

brumster, 2018-06-19:
I fear I might be walking down a hardware issue here, which I don't really want to get involved in resolving, but I'll play along for now... but please keep tickets in GitHub related to code issues....

After you've flashed it and it restarts, do you get any windows USB connection notification/noise?
What version of Windows?

PeterSpam, 2018-06-19:
Thank you for your fast response. I get a notification that an unknown device was plugged in and in the device manager it is listed under "usb". I'm using windows 10 and don't believe that it's a hardware defect because I used this pro micro for previous relative complex projects and everythink worked.

brumster, 2018-06-19:
Cool, ok, so - when it's flashed and plugged in, ignoring the device manager for a second, if you go to USB Game Controllers, do you see the EDTracker listed there as a joystick?

PeterSpam, 2018-06-19:
Unfortunately not. It's just showing up as unknown device and I can't connect to it in the GUI-configurator. It's not responding at all and doesn't have an open Com Port anymore...

brumster, 2018-06-19:
It's definitely the catalina bootloader on a 32K, 16MHz ATMEGA32U4-based Pro Micro right? The XTAL is definitely 16MHz, and you've chosen in the Arduino IDE the "EDTracker" device type, and the "9250" variant before compiling and flashing?

You've installed the Arduino drivers I assume - since otherwise the bootloader wouldn't be detected and you probably wouldn't be able to flash it via the Arduino IDE anyway :) ?

PeterSpam, 2018-06-19:
I'm sorry but I don't know the bootloader version. My pro micro is the 3.3v 8MHz version and I'm using the newest SparkFun driver. Also I tried to flash it as "EDTracker 9250" and "SparkFun pro micro 3.3V 8MHz"

PeterSpam, 2018-06-19:
Thank you very much!!! After the few changes it works now. I'm very happy about your quick response.

brumster, 2018-06-19:
Cool! Will close as resolved...

closed 2018-06-19 by brumster

tklenke, 2018-12-07:
Hey!  That worked!  It might be nice to add some more explicit instructions for 3.3v boards in the notes.  Thanks!

dogledhan, 2020-01-23:
Hi... I'm still struggle here, having same issue.

In this case, I use mpu6050 and pro micro 3.3v

How to reset the arduino (or any other solution)? Tried connecting rst+grd, the com&port showing for a while and then back to unknown usb device.

brumster, 2020-01-23:
As above, the code is for 16MHz Pro Micro's. The 3.3v ones are typically 8MHz devices. You can hack it to work (see the previous comment above about boards.txt and refresh rate) but your mileage may vary :) !

dogledhan, 2020-01-23:
Hi, thanks for quick reply. I really appreciate it

I've succeeded flasing, now another issue. When I calibrate the board by flasing it (EDTracer2Calib 2.5.3), the device not recognizing again. How I set the f_cpu value on the EDTrackerUI 4.0.4.0??

Is there any other way to calibrate without flashing it?? 

sorry my English not good.

brumster, 2020-01-24:
Hiya, no problems ;)

F_CPU is nothing to do with the EDTrackerUI, it's a value that defines the CPU speed of the AVR. It's part of the Arduino toolset. You need to edit a file :

<install location of arduino software>\hardware\edtracker\avr\boards.txt

Whatever device you're using (probably EDTracker6050) there is a build.f_cpu=16000000L line, Change the 16 to 8, rebuild and reflash via the Arduino UI.

The EDTracker UI only flashes 16MHz devices. Because you've used a 3.3v 8MHz you're going to have to custom build the firmware using the Arduino IDE.

dogledhan, 2020-01-24:
Hmm, got it. My module is working now, the only problem is the yaw drift (because I'm not calibrate it maybe). If I use mpu 9250 (cheaper than 9150), there is no error and no drift right?

brumster, 2020-01-24:
The yaw drift will be unbearable on any device if it's not been calibrated :) but on a 6050 it would be particularly bad. Just build both firmwares for your 8MHz, flash in the calibration firmware first, connect and calibrate, then once that's done flash the main firmware in.

Or, as you say, use a 9150/9250 (they are much better, in all fairness, with the mag)

dogledhan, 2020-01-24:
Another problem. When I plug my arduino in DCS World (not even run opentrack), why I can't steer my plane?? Work perfect in ATS

brumster, 2020-01-24:
I'm sorry, I'll have to suggest we stop using this old closed issue for this discussion.

If you have a defect/issue to report with the code, please create a new issue, however it looks to me like you have not followed the instructions correctly and are not using the custom device. Please read the instructions carefully, both here on GitHub and also at http://www.edtracker.org.uk/index.php/diy/flash-it/arduino-ide

Apologies but it's the proper thing to do rather than clutter up an old, closed issue ;)

dogledhan, 2020-01-24:
I'm so sorry... This is my first arduino project. I'll create new issue soon.