brumster/EDTracker2 #9: Compatible sketch for NodeMCU 8266

cazzoo, 2018-07-13:
Hello there!

I'm new there on head tracking and also on micro chips programming.
I am wondering if we would imagine switching from Arduino to NodeMCU using ESP8266 or ESP32.
This would be really neat since these boards have wifi connection so we could imagine having wireless head tracking system (if couple with some sort of batteries).
Do you think the actual sketch (designed for Arduino) would work as is with NodeMCU ?
If no, do you think changing to that would represent a huge amount of work ?

Finally, if my assumption using such board is wrong (because not powerful enough or whatever), what would you suggest as an alternative ?

Thank you btw for the initiative and for the sharing (big cheer).

Kindly regards,
Casimir

cazzoo, 2018-07-13:
Maybe this question could stand in the issues of the hardware repository : https://github.com/brumster/EDTracker2_ArduinoHardware

Sorry if it is the case

brumster, 2018-07-14:
ESP8266 is for 802.11 wifi; for connecting to networks and doing TCP/IP stuff. It's not really for proprietary wireless comms between two devices, like a head tracker and a dongle. For such an approach to work, you're really changing the entire architecture of EDTracker... you would be making a wireless device that connects onto your LAN and then transmits the head positioning information to another device - most obvious suggestion would be to do UDP broadcasts to your PC, and have something like Opentrack listen and process the packets accordingly.

The sketch will not work with NodeMCU, no. The entire USB stack would be unsupported, I believe, among other things.

It's an idea, but it's vastly different to the current architecture (which makes EDTracker look like a USB joystick), so would be a fair bit of work for someone.

JabberwockPL, 2018-09-12:
ESP32 supports HID devices over BLE, so it can be seen as a wireless Bluetooth joystick. That, I suppose, would require less changes in the architecture?

I have managed to run MPU9150 sketches on ESP32 and they generally work, except I cannot figure out the problems with yaw...

brumster, 2018-09-14:
In the sense of the original question, I am going to close this. No plans to re-architect in order to support NodeMCU 8266, no. ESP32 is represented on a separate ticket.

closed 2018-09-14 by brumster