brumster/EDTracker2 #10: Startup failure only on a "cold start"

manoukianv, 2018-11-28:
Hi,

First of all, thanks for this amazing EDTracker !

i have an issue on start EdTracker with 4.0.5 and 4.1.0
It's happened only with a "cold start" : when i flash the firmware, the edtracker is detected and can be monitored, but, il i plug the device a second time, it doesn't work.

the startup is failed because the 9250 never trigger the INT, so a unlimited while wait...
```
  //Loop until MPU interrupts us with a reading
  while (!new_gyro)
    ;
```

i update the 4.1.0 version with a "dirty" delay in the setup method to wait that all the device are "wake" and it fixed the issue : 
[image]

I hope this will help.

Vincent

brumster, 2018-11-28:
Hi Vincent, many thanks for the report. I suspect some devices (MPU boards) show different behaviour or it might be down to individual builds, but there probably needs to be a delay after the MPU enable call. Could you do me a favour, if this is very repeatable, can you reduce down your 5 second delay in increments and see what the minimum necessary delay is? Maybe try 50ms, 100ms, 250ms, 500ms, 750ms, 1s... and so on...

If we can determine a safe delay to introduce into the MPU enable then I will include one for edge cases like yourself but don't want to introduce a 5 second delay to startup if we can help it ;)

manoukianv, 2018-11-28:
Yes i think startup "boot" delay is required. I have a logical analyzer, so i will track and share with you the startup signal from the stock firmware and one with the smallest delay.

manoukianv, 2018-11-28:
I found the answer in the 9250 datasheet : 100ms Max to wait from power
[image]

brumster, 2018-11-28:
Yeah, that's built in to the Invense motion driver though, in the mpu_init() function ;)

Are you saying a 100ms pause prior to the initialise_mpu function in setup() sorts the problem for you?

manoukianv, 2018-11-28:
i'll try in few minutes :)

manoukianv, 2018-11-28:
I read the mpu_init() code : it send an reset command on i2c channel during the init command.
This is probably the issue, go on analyzer to check that ! :)

manoukianv, 2018-11-28:
ok let's see the first I2C communication on the sda line
[image]
[image]
First screen shot is fail start and second one is ok start.
The errors occurs in the mpu_init 
[image]
in the mpu_init, the arduino send a reset message to the 9250, so it send : 

- send a message on the bus to init comm with the mpu (which have address 208) and wait ack
- send a message to set power_management in mode 1 (send the value 0x6B) and wait ack
- send a message to reset confif (send message 0x80) and wait ack

During the fail start, the 9250 doesn't answer to the first i2c message, it does not provide ack, so the arduino try to make a second call 38ms later with the same issue, and after that init sequence is brake and the mpu is not init and setup. 
On the successfull start, the mpu return ack on all i2c message.

manoukianv, 2018-11-28:
Ok i try various delay time : 100, 200, 300, 350, 400 and 500.
From 350ms it's start to works, 100% of successfull start from 400ms. In code i put 500ms to add security.
I put the delay(500) just before the "initialize_mpu", i think it's better.

brumster, 2018-11-29:
Great stuff, many thanks for your efforts! I'll adopt this in. Just to explain my thinking for the record - obviously the majority of people don't see this problem but there does seem to be a small portion of DIY EDTracker builds out there in the wild that exhibit strange behaviour when built with MPU9250s. While I can't prove it, I suspect there are slight differences in quality control between 9250 devices or breakout boards :). I think a 500ms delay incorporated into the startup isn't going to cause anyone an issue ;)...

closed 2018-11-29 by brumster

manoukianv, 2018-11-29:
I don't use breadboards or else, i sold wires beetween Arduino and MPU : best way to try.
I'm agree with you for the 9250 quality...

commit 63b66b3d (2018-11-29): Merge pull request #11 from manoukianv/patch-1

Add wait time in setup for MPU9250 boot - Fix #10 issue
commit d123f58c (2018-11-29): Add wait time in setup for MPU9250 boot - Fix #10 issue

add a wait time, in the setup, to be sure that the MPU9250 can answer I2C bus after "cold" startup.