brumster/EDTracker2 #12: fix too many arguments to function

privategrim, 2020-01-25:
Hi, I got this error when I try to uploading the calibration sketch. How I fix this? I'm prretty sure that I follow the instructions correctly.

**mpu_set_accel_bias_6050_reg(aBias, 1);**
too many arguments to function 'void mpu_set_accel_bias_6050_reg(const long int*)'

brumster, 2020-01-26:
Ah looks like the function changed in the Invensense drivers - try removing the second variable (so just leave the aBias parameter; drop the "1"). Search for all calls for mpu_set_accel_bias_6050_reg and change them all, so there's only the first parameter passed. I suspect it will now compile? If so, I'll get the fix uploaded... cheers.

privategrim, 2020-01-27:
Are you saying to delete "1"?? When I delete 1, it can be uploaded. When I calibrate via EDTracker UI, the two dots keep moving around and can't be centered (tried many clicks on "calculate bias value"). But, after I test on DCS, the drift is drastically reduced.

I think the drift is bearable for me. But soon I will replace my MPU6050 for better result.

In EDTracker UI, I let it stay still for about 15 minutes, and than I click "save drift compensation".

brumster, 2020-01-27:
Cool, ok, I will update the code to fix the incorrect function calls in the next few days but for now you have fixed it locally yourself.

privategrim, 2020-01-27:
Ok... Thanks for your assist, appreciate it.

brumster, 2020-01-30:
Fixed in f50a470 commit

closed 2020-01-30 by brumster