brumster/EDTracker2 #13: MPU-9250 appears to crash/freeze after brief use

butchflowers007, 2020-01-31:
I have an Arduino Micro ATMEGA32U4 and a SparkFun MPU-9250. I have the latest firmware flashed and I am trying to use the EDTracker software to calibrate the magnetometer.

After a couple of movements, the program appears to freeze and the Yellow LED on the Arduino board will stay solid Yellow or go dark entirely. I have observed the same behaviour without the EDTracker software running.

Letting the board rest on my desk doesn't result in any issues, its only when I start moving it does it appear to drop out.

I am not sure how to provide debug info, but happy to provide it once I know how.

brumster, 2020-01-31:
Hi there; never had any reports of this as a code issue, sorry. I think it more likely to be a hardware issue - check cables, soldering, visually inspect the board/etc it might be a 'marginal' connection somewhere.

You can recompile the firmware with the `#define DEBUG` uncommented, then you will get some additional output on the Arduino serial monitor (under Tools menu) but I suspect it won't tell you much more. Still, worth a try!

butchflowers007, 2020-01-31:
Thanks brumster! I have been doing some investigating and I think its an i2c issue. I don't have any pull-ups on my SDA and SCL lines. My Arduino is 5V but the MPU-9250 is at 3.3V, I was relying on the Arduino's internal pull-ups to do its job, but maybe the difference in voltage is the cause. I will fiddle around and see if adding external pull-ups will help.

Not sure if its any use, but when the device freezes the SDA or SCL lines will be at 3.3V or 0V.

brumster, 2020-01-31:
An i2c freeze/hang does sound a likely candidate... your approach sounds sensible! Give it a try ;)