brumster/EDTracker2 #14: Losing Arduino Leonardo COM port

weepop, 2020-03-02:
Can anyone tell me why I lose my COM port 6 entry for my DIY EDTracker  GUI when I turn off my PC.

To be more precise I only lose it if I restart my PC with the EDTrack USB plugged in.
If I wait till PC is up and running before I plug it in it does not happen,the entry is still there.

I think I should not have to do this and would appreciate any help to stop having to.

regards weepop

brumster, 2020-03-02:
When you turn off your PC, is power to that USB port turned off as well or does the EDTracker remain powered up?

weepop, 2020-03-02:
It makes no difference ,It will fail if I leave PC powered when off , or if I completely isolate the PC at the mains plug.

brumster, 2020-03-03:
Ahhh ok, sounds a bit like these reports maybe :
https://www.reddit.com/r/EliteDangerous/comments/3f4tq1/edtracker_win_10_update_solution_found/

https://forums.frontier.co.uk/threads/edtracker-not-working-in-the-game-in-windows-10.261009/

If so, this was an issue with the Arduino drivers and Windows 10, I'm not aware it ever got fixed. Make sure you are using the latest Arduino drivers (they are packed in with the Arduino IDE in a 'drivers' subfolder)... however I cannot explain how the physical unplugging of the device is the fix when totally powering the PC off at the mains doesn't, sorry :(

weepop, 2020-03-03:
Thanks for that. 
I am using windows 8.1 not 10
I think I will just live with unplugging the EDTracker when I turn of PC ,didnt like it sitting there with its LED glowing anyway so I will plug back in after booting.
weepop

closed 2020-03-03 by weepop