brumster/EDTracker2 #16: Getting 2 axes only from MPU-9250

MiOiM, 2020-05-17:
Hi guys.
I'm building 2 EDTrackers with Pro-Micros and MPU-9250 (the ones with 10 connectors) following the Bartybee guide, the 9250 is on top of the Pro-Micro, but I've turned it 180 degrees horizontally as I've noticed that the INT and GND pins match exactly pins 7 and one of the GND pins of the Pro-Micro, like this:
 
[image]

So, I have built 2 of them.  Flashed and calibrated, the magnetometer looks like this:
[image]

However, while one of them seems to work fine, the second one is displaying Gyro data on only 2 axes in the EDTracker UI tool, while the 3rd one just changes between 0 and -2. Also, when trying to move it on one axe only, the pitch axe for example, the head moves from the lower left corner to the upper right one.

I have reflashed it, restored to factory defaults and calibrated multiple times (the magnetometer screenshot is from this one) and checked the wiring but I cannot see any change. I have calibrated the other unit a few times and is still fine. They are never connected at the same time.

[image]

Do you think I have a faulty MPU, or this particular issue can be due to incorrect wiring? How would you troubleshoot it?

Thank you for your help!

brumster, 2020-05-17:
Hello there! This is really intended for issues with the source code but I'll play along for now ;) !!

If you can capture some of the debug output from the Arduino IDE it should be possible to see the raw values for each sensor axis. But the code is rock solid in this regard, I'm fairly confident there are no issues around axis reporting so it's most likely an issue with build or config. Try another orientation setting; what happens? Did you follow the initial commissioning instructions in the user guide (ie. after initial calibration, click "restore factory defaults" then perform calibration)?

MiOiM, 2020-05-20:
Hi @brumster. 
Thank you for your help! I kind of knew that this is not the best place to look for help, but I didn't see much movement in Reddit and wasn't sure if opening a ticket in the main EDTracker website, as this is not a EDTracker Pro related issue...
Please feel free to tell me a better place to go crying for help ;P

So, I have tried to load the project in Arduino to give you the debug output. 
Saved the 'edtracker' folder from https://github.com/brumster/EDTracker2_ArduinoHardware into my Arduino\hardware folder, but Arduino didn't find the resources when compiling EDTracker2_9250.ino. So I copied the 'edtracker\avr folder into \Arduino\hardware\arduino as there is a avr folder there...
After restarting Arduino IDE, I can compile it with no errors but a good amount of warnings. 

But when trying to flash it, I get this error:
`java.io.IOException: Cannot run program "{path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2047)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
An error occurred while uploading the sketch`

I don't want to be a pain, honestly. I understand the code cannot be the problem. Soldering looks ok, and I think my wiring diagram is correct. Is it possible that having both modules so close (a bit less than 5 mm between them) is causing problems?

I am considering rewiring it completely, maybe following this time the same position of the modules in the instructions provided on the website, as mine is rotated 108 degrees horizontally.

MiOiM, 2020-05-22:
Hi @brumster 
I have dismantled it and connected on a breadboard. Also flashed EDTracker2Calib 2.5.3. Gyro Z is still static, even though X and Y seems to work fine. The white dot is permanently located in the lower-left corner and tweaking Gyro X and Y values I have managed to place the yellow dot in the bull's eye. 

The MPU module is positioned horizontally, with the soldering holes closer to me. When moved:
- Pitch is working correctly, the yellow dot goes up and down.
- Yaw does nothing.
- Roll moves the yellow dot left and right.

All three Accel values react with movement. When left static for a few seconds, they stay in values around -3270, 1300 and -75. Tried to tweak them but I don't think I can get them anywhere close to 0.

Does all this make any sense? 
Thank you, brumster.

brumster, 2020-05-23:
Hi MiOiM, yes, it sounds to me like you have a faulty MPU. While the 2.5.3 calibration sketch is really meant for the 6050 devices, it can be used to diagnose as you have done. It sounds to me like the Z gyro is just dead, not reporting anything. Your mag is fine, hence the "sphere of blobs" looks good, and you're getting pitch and roll report, but the yaw is not playing ball. The fact that you built 2 identically at the same time, and one works while the other doesn't, is telling of a hardware problem.

The MPU-9250 chip from Invensense has never had the best quality rate in our experience. It's part of the reason why we dropped it on the EDTracker Pro and went for a separate HMC mag so we could keep the good old MPU-6050 :). Try swapping the MPU-9250 board between your two devices you build (if you can) and I bet the problem follows the MPU-9250.

Phrank916, 2020-05-28:
I've just built my DIY EDTracker and had a similar experience. I also wired everything in a similar fashion to MiOiM with the MPU spun 180. I used strip-board to sandwich them together and it just makes sense with the pin 7 and ground lining up, then a simple crossover of pins 2&3. After struggling to calibrate a few times and getting very crazy response, I also loaded the calibration sketch and confirmed a dead Z axis on the gyro. So, Amazon return engaged and I ended up ordering a 3-pack of the 6050s. One of them is BOUND to work, yeah?! I'll follow up if neither of you mind my thread hijack! 
BTW, is there a possibility of getting a peek at a schematic for adding that HMC mag you mentioned above, or is that a "Pro" exclusive? If not, I can live with a bit of drift to avoid the 9250 quality control headaches. Cheers! 
[image]

brumster, 2020-05-28:
Yes, the HMC mag is a feature on the EDTracker Pro not DIY, it's a commercial product.

Phrank916, 2020-05-31:
I got the 6050 board in and it works great now! The drift is noticeable but I plan to try calibrating again and hopefully get it working better. I moved the button placement and trimmed the strip board down in size, and then designed and printed a custom case. It's a bit rough on the first go because I have a 0.8 nozzle on my printer but it's serviceable. 
[image]

On a side note: Is there a better way I can contact Brumster (Dan)? Since it's indicated on the website that the Pro won't be re-stocked, I wanted to ask about the possibility of doing my own modifications to the firmware for adding an external AK8975 (since that's what's on the "2 die" IC of the 9150). I figured the GitHub "issues" forum isn't really the place to discuss. 

If I were to submit a trouble ticket with my direct email is there a chance you might respond and be interested in a SHORT "non-public" conversation regarding the above? I promise it wouldn't be a mundane or annoying conversation asking for loads of guidance. I'm an EE with a background in embedded code and hardware and have done quite a bit of "hobby" AVR projects (even before Arduino became synonymous), so I know it's something I can do. I do realize I could simply fork the repo and get to work, but I suppose I'm just wanting your blessing. I also foresee an issue interfacing with the "binary only" GUI.

brumster, 2020-06-01:
Hi chap, I'm no longer involved in the EDTracker company, I left about 2 years ago now ;)

But just fork the code, honestly, no-one is going to take issue - that's the whole joy of GitHub and an Open Source license. The software is a problem, you're right, and that's not something I was involved in upkeeping as Rob did all of that. I'll have a conversation with him about what we can do, but the serial communications between device and UI is pretty simple, you should be able to deduce it from code or we can put up a guide... leave it with me to discuss with him.

MiOiM, 2020-06-02:
Hey @brumster, just to thank you for your help.
Should I close the issue?

brumster, 2020-06-02:
Yeah let's close this one; you can always raise new ones for separate questions/etc... best way to go, keeps the issues clean and on-subject ;)

closed 2020-06-02 by brumster