brumster/EDTracker2 #17: Licensing

joshuahant, 2020-08-10:
Hi Dan,

Is there any chance we could see this software get a license? As it stands you hold the copyright which is fine, but also if someone wanted to fork this and add changes on that fork then they would potentially be opening themselves to legal problems. With a more permissive license, others can keep maintaining the project without worry and you won’t have to be the only one updating it. Just wanted to see what your thoughts are about this.

Best,
Josh

brumster, 2020-08-10:
Hi Josh,

For the portions of code that are/were "ours" (ie. what Rob and myself original came up with), it is all under the MIT Open Source licenses which allows people to freely modify, distribute and use so there should be no issue with forking/updating. Obviously there are portions of the repos that leverage similar open source agreements from other parties (libraries/etc) which we honour but for the code that we wrote, there is no issue - fork it, do what you want with it, just honour the license by passing that freedom to others (ie. never close-source it). Rest assured if you fork the repo and look to maintain a separate stream of EDTracker, there will be no issue from myself ;) I can't do this forever :D !!

@pocketmoon any issues you foresee?

brumster, 2020-09-24:
Hi Josh, can this be closed now? Or if not, can you highlight/explain what portions of the license you feel are restrictive?

joshuahant, 2020-09-24:
Yes that's a good enough answer for me. Thanks!

closed 2020-09-24 by brumster