brumster/EDTracker2 #23: Firmware not persisted on device?

liwp, 2020-11-12:
I suspect this is my fault rather than a problem with the firmware or device...

The USB port on my original arduino came off (I should've listened to the warnings). I bought a new one, but I didn't have the tooling to remove the original arduino from the breakout board (I've got the Hobby Components kit), so I soldered the new board on the same breakout pins on top of the original arduino.

Now, I can load the firmware on the new arduino and everything works fine. But when I disconnect and reconnect the USB cable I have to reload the firmware again. It looks like the firmware isn't persisted.

I suspect this is because of my hacky soldering. Maybe there's a read-write pin on the arduino that's always on because of the second defunct arcuino in the circuit? But I don't understand enough of what's going on to be certain.

Anyhow, thank you very much for all the work - EDTracker has been awesome!

brumster, 2021-02-27:
Not really anything I can do here, I'm afraid. I'll put it to community support but this is unlikely a code issue, most likely a faulty Arduino/Pro Micro hardware, sorry.

liwp, 2021-02-28:
No problem! I was just hoping that someone could explain to me what might be going on, since I don't understand enough about how everything works to work out if it's faulty hardware, something in the software, or something in my soldering.

karlo922, 2021-03-01:
Maybe you reached the "reset" port while soldering with some spare solder? So you maybe "reset" the device?

brumster, 2021-03-01:
I've never seen this done before, having two Arduino Pro Micros on the same header pins will almost certainly cause some dodgy behaviour if you haven't disabled the old one (albeit it's got a USB port broken on it). i2c pins shared, the old firmware will still be in the old Pro Micro "doing it's thing" while your new one is trying to access the same bus too. Best advice I can think is break the VCC tracks on the old Pro Micro board so that it's MCU is totally 'dead'. That assumes you can still get to them, which you probably can't anyway :).

liwp, 2021-03-01:
Thank you both! Probably wasn't the best decision on my part, but I got frustrated after having tried removing the old board from the headers without success :slightly_smiling_face: 

Put some hot glue on you USB ports kids!

@brumster feel free to close this issue.

brumster, 2021-03-01:
No problems, will do ;)

Solder wick FTW, kids!

[image]

closed 2021-03-01 by brumster