brumster/EDTracker2 #24: EDTracker website

CommanderCaveman, 2020-11-21:
Hi
Didn't know where else to post this so dropping a line here.

I used the WayBack Machine [https://web.archive.org/web/20201108130443/https://www.edtracker.org.uk/](url) to retrieve information on the edtracker, its not totally complete and I had to go back through the various captures to gather the majority of the websites downloads, I think its all there over time.

Its a shame it appeared to all go wrong so quickly for the team.

miniskipper, 2021-01-04:
I am also very sad. This is a neat little device that just works. BUT: as a Linux user I always had to use a Windows VM to calibrate the thing and change settings.
Maybe there is a little chance, the code of the GUI could be made open source now? Would love to build an alternative utility for Linux/Mac users.

brumster, 2021-01-04:
Firmwares, update tools and binary builds of the GUI are now uploaded into this Git repository (under [release](https://github.com/brumster/EDTracker2/tree/master/release)) as a replacemnt for the now-stood-down website. Unfortunately I don't have the source code to the GUI and the owner does not appear to be interested in releasing it, so most likely it would need volunteers to write a new one...

closed 2021-01-04 by brumster