brumster/EDTracker2 #25: No interrupts from MPU

FalcoGer, 2021-01-22:
Hello.
I have built this device and I have messed up the pins on my PCB.
Arduino Mini + MPU9250

FSYNC to GND (instead of floating), as I've read this in the MPU refsheet and I have swapped SCL and SDA pins.
I connected Interrupt on Pin 7.

Unfortunately I was an iditot and soldered the boards straight onto the PCB without testing. I'll certainly not make that mistake again.

In software I have set the pins as i have them physically.
I have debugged the software and I am stuck in the loop waiting for the MPU interrupt. I have also checked the interrupt pin on the board and there is nothing there, even if i move the device around.
I have measured that there are ellectrical connections between all the pins I have set up and my soldering work seems to be good.

I have also measured the I2C bus and there is traffic (2 packages) just after uploading the firmware, but nothing else after that (since the software is stuck in the waiting loop, I imagine).

I'm also bit confused by how the I2C traffic is generated in the first place. I tried to trace the function calls but there is no digital write or wire calls anywhere. Also the interrupt handling is set up with INT6, with a rather weird comment saying 6 for 6 etc. I'm not sure if that's the pin number for which you want the interrupt or what?

I'd rather not unsolder everything again, do you know anything that's obviously wrong? Will the mpu not work with fsync tied to ground?

brumster, 2021-01-22:
First, is it REALLY an Arduino Mini or do you mean a Pro Micro ;) ?

FalcoGer, 2021-01-22:
apologies, I mean a pro micro.

brumster, 2021-01-23:
No worries on the Pro Micro/Mini, just wanted to check before we burned away a load of time trying to get a 328P working ;) !!

I'll upload a diagram into the Wiki later (I don't have KiCad) but basically looking at your diagram on there, it looks fine (I assume the ground plane/tracks aren't showing but are all good!). Your connections match the common layouts for Pro Micros and MPU92x0 boards. The tying of AD0 (the pin next to INT on the MPU) is a recommendation but in reality I don't think it's an absolute must.

You can't move the pins around for i2c if you want to use hardware serial from the microcontroller. I think there are software-based implementations of i2c out there but we never used them in EDTracker. I would image for reasons of performance but also memory consumption on the device; we're right up to the limit on the Pro Micro ;)

The older MPU6050 sketch used to offer polling or interrupt based comms with the MPU. It's still in that codebase, but it was removed from the 9000-series because, again, we were struggling to get the code size down the mag related stuff and some things had to go... polling support got pulled. You could always manually edit the code to re-enstate it and loose your interrupt-driven approach if it was absolutely necessary, but what you've drawn there should work.

FalcoGer, 2021-01-23:
Should FSync be tied to ground or not?
Also what you're saying is that I have to scrap the boards or cut the tracks and hotwire something?

brumster, 2021-01-24:
Regards FSYNC, ultimately follow the datasheet of the MPU-9250 IC - which says yes, connect to GND if unused.
Orientation of button is not important
If you've had boards made where the connections are incorrect, you can indeed cut tracks and use mod wires to 'fix' them.

FalcoGer, 2021-01-24:
I desoldered the board and will test on breadboard next before ordering a new one. Thanks for the help.

closed 2021-01-24 by FalcoGer