brumster/EDTracker2 #26: Arduino Leonardo + MPU6050 Usb device not recognized

Fratee, 2021-01-25:
Hi,

First thanks for you dedication to the project, having moved everything here to make it available to everyone, really appreciated.

Anyway I have a strange issue i couldn't figure out. 
my setup is an Arduino Leonardo + MPU6050. I'm able to load the calibration sketch and calibrate the device.
but when i load the final firmware Windows says "USB device not recognized".
In the device manager i see:
[image]

I tried to lower the DEFAULT_MPU_HZ to 100 as suggested in an older issue but no luck.

Any idea on the issue?
(Also i made it work a couple of years ago with this identical setup, now not working anymore...)

Elia

brumster, 2021-01-26:
Hi Elia,

Windows drivers? Maybe try USBDeview tool to investigate what drivers are recognised when the device is plugged it; it can also be used to remove the device entries from the registry and clean up your PC.

Try a different cable too just to be safe, and a different port in case it's a power issue or USB3 driver related.

Fratee, 2021-01-26:
Thanks for your reply.

Here USB deview infos:
[image]

Changing usb port to 2.0 and also cable did not work.

By "cleaning" do you mean removing the reg entry?
[image]

The thing is that the calibrate sketch is working, do you know changes between the two (dev HID maybe?) ?

Elia

Fratee, 2021-01-26:
Good news, managed to fix.

Here are the changes i made to `EDTracker2_6050.ino` :
added delay after serial begin
```
Serial.begin(115200);
delay(500);
```

commented:
`//TWBR = 12; // 12 400kHz I2C clock (200kHz if CPU is 8MHz)`

changed port from D to C:
```
  // Disable internal I2C pull-ups
  cbi(PORTC, 4);
  cbi(PORTC, 5);
```

Again this is now working with Arduino Leonardo with the MEGA32U4 and MPU8060:

Here my edited file:
[EDTracker2_6050.ino.txt](https://github.com/brumster/EDTracker2/files/5873164/EDTracker2_6050.ino.txt)

You may consider changing these also in the repo, so it reflects the calibrate settings. Maybe? dunno

Thanks for your support @brumster , you got me thinking a little 😉 
(you can mark as closed if you are satisfied)

Elia

brumster, 2021-01-26:
Ah good spot on the i2c pullups; I will do some sanity checking between the two sketches and update accordingly! I'll keep open for now just to ensure I don't forget, as I might not get round to it until the weekend ;)

brumster, 2021-02-28:
Fixed in 2.21.1

closed 2021-02-28 by brumster

andywong1026, 2022-06-06:
Very  good. Please post the firmware file in .hex format so I could flash with xloader.

brumster, 2022-06-10:
Done - now available in the firmwares folder