brumster/EDTracker2 #27: Mag oops (7)

CommanderCaveman, 2021-01-31:
Pro micro clone with 9250, wired up direclty wihout interface board (to make unit more compact) and without reset button (I understand that rst button is not required for the 9250).
The device is recognised by the PC (after moving to motherboard USB socket, case sockets caused issues including bricked first attempt)
Uploading appears to have been competed. (I have not tried uploading the calibration sketch (only just discovered a reference in one of your YouTube videos).

Serial monitor in IDE is returning Mag oops (7), my power led and Tx led are constant and Rx led is flashing rapidly. In the GUI it doesnt connect but sees that there is a Leonardo on my Com11, it starts calibration but not let it complete as nothing was appearing in GUI (not sure if it should).

Do I suspect corretly that the 9250 is kerput?

CommanderCaveman, 2021-01-31:
I have now replaced the 9250 with a spare, it is now connecting within the UI. I am not getting any reading for Yaw or Yaw drift, I am getting movement of the head in a nodding fasion around the y axis, with USB to the right, but no movement around the Z axis, ie looking from side to side. I have run the mag calibration at a high sensitivity, and the graphic shows a really tiny green head with a single red dot a long way away.

I see nothing in the serial monitor in the IDE, do I need to enter the DEBUG line in rather than uncomment it, I could find the #define DEBUG code but nothing more, my C is so rusty it resembles ore.

brumster, 2021-02-01:
You're following this guide, roughly, right?
https://github.com/brumster/EDTracker2/blob/master/docs/EDTracker%20User%20Guide%20(Magnetometer%20Versions).pdf
You've calibrated the mag?
Also since you've done the build yourself you may have to experiment with the orientation options to find one that relates to your MPU layout.

CommanderCaveman, 2021-02-01:
DOH! I hate children that dont sleep and I am trying to do this whilst tired, I was holding the unit still to calibrate the mag. This is the best I have managed to achieve.
[image]
Good reminder about orientation,my 9250 is 180deg out from the design reference.

Its too late to try out in game now, but it looks like I am now on the straight and narrow.

Thank you for your kind and helpful support it is very much appreciated.

Now I wonder if this can be reproduced using the new Pi nano.

brumster, 2021-02-02:
Yup that looks better. Movement needed for mag calibration; it needs to be still for the gyro auto-bias only (when it starts up or when you manually kick one off). I'll close ticket for now' re-open if you need to ;)

closed 2021-02-02 by brumster