brumster/EDTracker2 #28: Sketch installed by HobbyComponents

GMcDowellJr, 2021-02-19:
I’m trying to us EDTracker on a PS4 by routing it through a Titan Two (T2) from consoletuner.com. I had to write a script for the T2 to calculate angular velocity from the XYZ angles to mimic the data the DS4 controller sends. I also had to work the angles to mimic accelerometer readings. It works but there are a few challenges that I think are a result of the autocentering in the sketch.

When I get close to the middle of the screen vertically OR horizontally (meaning it happens away from the established center) my view snaps to the vertical or horizontal line. Likewise when I move off the line there it feels “sticky” until I snap off the line. Does that sound right?

I think those snaps, which are short, but quick, movements, are being seen by my script as a much faster move than intended and I end up overshooting and stuck out away from the center forcing me to turn off motionlook for a moment to recenter. I think turning off autocenter might fix this issue.

All that leads to my real question, is the sketch for the 9250 here on GitHub the same (or likely the same) as what HobbyCompents would have flashed? I’m okay messing with the codes assuming I can go back to a default state by reflashing with the original sketch.

And, assuming I can get my EDTracker back to where it is today, I’m also going to see if I can just pull the accelerometer and gyro values from the 9250 and feed a version of them back to the PS4 directly. Seems like a better choice than converting those same outputs to Euler angles and then back!

brumster, 2021-02-19:
Hiya - interesting use case! Sounds cool if you can get it working!

In answer to your question, all I can say is "I would *think* so" but ultimately you'd need to ask HC. If you plug it in to a PC the UI will report the version string on the device in the UI, so you can check if that correlates with the source code. It should say "EDTrackerMag V4.1.1" if it's this source.

On the 6050 sketch what you describe is indeed how the autocentre worked, it tweaks the drift compensation value but it only does this when you keep your head steady in the 'straight-ahead' position for long enough. We basically figured if people are holding a flat position looking at a certain angle for long enough, it's *most likely* straight ahead (that's just the nature of gaming) so it was safe to make drift calculations during that period. If you're head is tilted up/down then we don't. It should be fairly obvious in the source code ('self centering' comment).

GMcDowellJr, 2021-02-19:
Thanks for that. I did check with HC and it is the EDTracker2_9250.ino file

I love the idea of recalibration when centered. Before I got my EDTracker I was working with SmoothTrack.app (face-tracking with a phone) through opentrack and tried to implement something similar. Tried being the operative word - lol. Is snapping into and out of this zone expected? When I read through the code I thought it wouldn't start the calibration process until I had been in that zone for some small amount of time. In the 6050 code it's "ticksInZone >= 10" if I'm reading it right. I don't see something like that in the 9250 code. Is this the part of the 9250 that deals with is "newv[1] < 6000.0  && newv[1] > -6000.0"?  Interesting that it's only for one axis. newv[1] is pitch, yes?

Any thoughts or hints on pulling the gyro and accel value directly? HC pointed me to a sketch that deals with the IMU so I have a place to start... well, that and this code! lol 

Thanks for continuing to host and support this work! (now if we could just get pocketmoon to release the gui) :)

brumster, 2021-02-19:
We only do the 'drift compensation' concept in the 6050 sketch because it doesn't have a magnetometer, so it's got no clue what 'true north' is. Without any external 'real' reference, just a hyro and accelerometer, the 6050 chip has absolutely no way of keeping a measure of a true reference point in terms of orientation around the vertical axis, the one axis that is the problem (because gravity is pulling straight down along it). So the idea we had was, well, let's take a chance that whenever it's level (we can measure that accurately because the 'tilt' forwards and sideways axes are, when level, 90 degrees to gravity) and we're not seeing movement, it's most likely straight ahead, or very close to it. So if the heading the DMP is reporting is off from our straight ahead position, we gently "tug it back" towards centre. But we filter it of course, we don't bring it in too rapidly, hence the 4 levels of auto-centering power, if you like.

On the 9250 as you correctly identify it works a similar way but we have the beauty of a simple mag reading to reference so we can measure how much the gyro drifts to the left or right of the 'reference' mag heading, and correct for it over time (again, heavily filtered because the mag can be noisy and also people are rarely in a perfect, consistent magnetic field - just moving your head in relation to your surrounding can drastically change the mag reading, so you can't rely on it - if you could, you'd just take the mag reading for rotational positioning of course - you try doing that and see the mess you get!)

I'm not sure why you'd want to pull the gyro or accel values (well, you could only remotely rely on the gyro really) but it depends how you configure the DMP, there are various options to configure it. Consult the Invensense Motion Driver 6.0 documentation for the various modes. We do bring back raw gyro and accel values in the call, plus the DMP output in quaternions, but we don't use the accel values and the gyro ones are only used to set the gyro offsets during calibration, after that they serve no purpose, we trust the DMP quats for positional information. But if you want them, for whatever reason, the raw gyro/accel values are right there returned on the call to **dmp_read_fifo()**.

Cheers,
Dan

GMcDowellJr, 2021-02-19:
Thanks for the explanation. Makes sense. I wonder if the 9250 was actually flashed based on that snapping effect I mentioned. Either way, I'm headed down the right path now. 

As to why I want the raw gyro; Elite on the PS4 can use the gyro in the DS4 controller to control headlook. The Titan Two device I referenced earlier sees the gyro and allows you to map another controller's settings over top of it. I had been trying to calculate angular velocity based on changes in yaw/pitch/roll relative to time -- and making some progress but it was rough.

I still have some tinkering to do with the T2 to get everything working correctly but, so far, just replacing newv[n] with gyro[n] here;

`    if (ii[0] > 30000  || ii[0] < -30000) {
      joySt.xAxis = gyro[0] ;
    } else {
      joySt.xAxis = joySt.xAxis * outputLPF + gyro[0] * (1.0 - outputLPF) ;
    }
      joySt.yAxis = joySt.yAxis * outputLPF + gyro[1] * (1.0 - outputLPF) ;
      joySt.zAxis = joySt.zAxis * outputLPF + gyro[2] * (1.0 - outputLPF) ;`

and remapping the joysticks to overwrite the gyro using the T2 seems to have done the trick!

I think there's a lot that can be stripped out of the code now and I need to use it in-game a bit to see what else might need to change but it's looking up!!

closed 2021-02-19 by GMcDowellJr