brumster/EDTracker2 #31: MPU 9250 calculating gyto bias forever #2

SPAZv, 2021-03-11:
I have this problem now, this problem was writen before here, but sketch with modified files is gone now because it was 3 years ago. Have someone working sketch for mpu9250? I use sketch from EdtrackerMaster but what i say gyro calculating forever.

brumster, 2021-03-11:
Sketch has not gone; the fixes were incorporated into the code on this repository. If you're flashing the latest source code and using the latest files off GitHub, then your seeing something else. Most likely hardware fault, reset button stuck, etc.

SPAZv, 2021-03-11:
Edtracker is working in widnows games controlers( but buggy because is centering yaw all the time but pitch and overaly thing is working, and working in opentrack but buggy because of centering) But dont work in Edtracker, before i made edtracker on mpu6050 and work perfecly (but yaw drift) so i bought mpu9250, and flash new sketch witch edtracker 9250 and now its not working in edtracker, i dont solder button, only 5 cables, vcc, gnd, scl, sda, int.

brumster, 2021-03-11:
"In EDTracker", I take it you mean the EDTracker GUI? Clearly it's working (to some degree) because you're getting a connection. If it previously had 6050 code in it you will need to click the "restore factory defaults" option before you calibrate it, assuming you've shared the Pro Micro and have just de-soldering the old 6050 and put a 9050 on it... ;)

SPAZv, 2021-03-11:
Yes, in Edtracker GUI 4.0.4 it show conected, external window "calibrating gyro bias" i going forever :d And without calibration done i can't click "restore factory default" because its grey button and cannot be clicked. And yes i desoldered mpu6050 and solder mpu9250. This is how it loks [https://imgur.com/dhJUjw2](url)

brumster, 2021-03-11:
Yeah, there's something not right with that I'm afraid. Maybe enable debug mode in the source code (uncomment the debug line), reflash it, then use the serial debugger in Arduino to connect and see what's going on. Probably a mag oops error, which means there is something wrong with the i2c bus or the MPU board... the quality control on these MPU-9250 boards does seem to be very 'variable', I've seen this and people have swapped a new 9250 in there and it's worked fine :( sorry, not much else I can add, the device is stuck in the startup loop... if you're happy you've flashed the latest code into it then not much I can suggest I'm afraid, sorry.

SPAZv, 2021-03-12:
In debug mode in serial port monitor it show Mag Oops 7

SPAZv, 2021-03-12:
Something werid happen, i switch back to working before mpu6050 and now its dont work, serial port monitor show nothing now, led on mpu6050 is working like before