brumster/EDTracker2 #32: Automatic yaw axis return center

Gaastragtx, 2021-06-16:
Hi, great project. It works fine but there is a problem...
 why does the yaw axis automatically return to the center position when I have to move the yaw axis?
In this way the view returns to the center but the head remains tilted.
Is there any way to avoid this?
Thank you.

brumster, 2021-06-16:
Hi there, this isn't the functionality/behaviour... possibly you have not calibrated the device correctly or you have set the orientation wrong? But no, the view should not return to centre!

Gaastragtx, 2021-06-16:
Thanks for the reply.
Due to the offline website the EDtracker software cannot connect and download the firmware from the website so I had to upload it directly with arduino IDE.
During the programming phase I received some error messages but the firmware has been loaded into the arduino and with EDtracker I see the movement of all 3 axes correctly. I tried to do the calibration but the yaw axis keeps returning to the center position. I use the 9250 sensor.

brumster, 2021-06-16:
Hmm, if you're sure you've assembled and orientated it correctly, and you're sure you've flashed the correct firmware into it and performed a successful mag calibration, then I can only suggest it's a fault IMU :(

MarceloF1000, 2023-04-16:
I have the same issue, but with pitch. It will always return to center. Yaw hols position just fine.

I've tried all rotation options. Mine is a MPU6050

I repeated the entire process and still have the issue:
- Erased EEPROM and flash with avrdude
- flashed the calibration hex file with avrdude
- used EDTracker GUI to calibrate. X, Y and Z values on green and bias values on 0. Dead on target. Accel values are all over the place and won't calibrate, but that should not matter.
- flashed the EDTracker hex file for MPU6050
- used EDTracker GUI to run and save drift compensation.
- tested with all rotation options and with auto-recentre off in the GUI

Any help would be appreciated.