brumster/EDTracker2 #33: Problem with the EDTracker GUI (HELP?!!!)

KeinEngelZlat, 2021-10-03:
Good time of day. I have a problem with the graphical shell of the EDTracker GUI: when starting the shell, an error appears "It is impossible to resolve the remote name: 'www.edtracker.org.uk' You will not be able to flash sketches to the device. Please check your firewall settings or internet connection."
My firewall is completely disabled, as well as the antivirus. When trying to connect to the site through several browsers, a message is displayed that the site is unavailable. Is there any offline version of the EDTracker GUI? Or maybe there is an opportunity to manually download the firmware for the EDTracker GUI and make the program see them offline?
[image]
[image]

brumster, 2021-10-03:
Hiya - unfortunately there's nothing I can do about this; EDTracker as a company folded a while back and the web servers have now gone, disappeared. On top of this, the GUI was not written by me and the source code was never shared, so the UI is no longer maintained... so I can't modify it to remove the flash functionality or re-code it to get the source images from elsewhere like GitHub... so unfortunately, you just need to ignore the error message.

If you want to flash binary images into your device without compiling from source code, the best I can offer is the command line flash utility:

https://github.com/brumster/EDTracker2/tree/master/release/updatetool

The binary images of the firmware are here:

https://github.com/brumster/EDTracker2/tree/master/release/firmwares

KeinEngelZlat, 2021-10-04:
Sorry, could you help me with the Update Tool? I have done everything that is written in the Readme, namely: 

> LAUNCHING THE TOOL
> 
> ================
> 1. Open the command prompt
> 2. Go to the directory where you unpacked the tool in step (1). for example
> CD c:\temp\updatetool
> 3. Run the tool with the following command :
> java -jar UpdateTool.jar <parameters>
> If you skip the parameters, the help text will be displayed, including a list of parameters.

As a result, after entering the command in cmd, the following inscription appears. Whatever startup parameters I enter, this text appears. Java is installed, I run cmd with administrator rights. Tell me, please, what could be the problem? Maybe some sequence of parameters should be observed? And as far as I understand, after performing these actions, the firmware should appear in the EDTracker GUI?
[image]
Java is installed, I run cmd with administrator rights. Tell me, please, what could be the problem? Maybe some sequence of parameters should be observed? And as far as I understand, after performing these actions, the firmware should appear in the EDTracker GUI? Or is the firmware being poured into the device itself via cmd? But then how do I calibrate the device if the EDTracker GUI won't work? Or after flashing the device in the program to decide and join?

And I apologize for my English, I'm writing through a translator.

brumster, 2021-10-04:
Hi, no problems :)

What version of Java are you running (java -version) ?
You *have* downloaded the rxtxSerial.dll file as well, right? This must be in the same folder as UpdateTool.jar

As you quite rightly say, this tool pushes the firmware into the device from the command line - it does not fix the UI problem. It is a command-line tool for flashing a firmware into the device instead of using the UI.

The EDTracker GUI will continue to work as normal - the only part of it that DOESN'T work is the "flash" feature. So you use the command line tool to flash the firmware into the device, and the EDTracker GUI for everything else ;)

KeinEngelZlat, 2021-10-04:
The Java version is shown in the screenshot. 
[image]

File rxtxSerial.dll also downloaded, located in the same folder as UpdateTool.jar . Everything seems to be right? And such a question back then: I use a bundle of Arduino Pro Micro + MPU 9250, is Pro Micro suitable for these purposes or is it better to replace it with Arduino Nano V3?

brumster, 2021-10-04:
Nano V3 is no good - it's not atmega32u4-based, it uses a different microcontroller, so the code won't work on that

On the java error, I've honestly no idea - I tried the update tool here and it's working all good on Java 1.8.0_301-b09 build. Maybe it's an Oracle JRE thing... presumably Oracle dropped a lot of the apache commons libraries from being bundled with their version? If that's the case, you maybe just need to bring those in (or use an older JRE might be easier!)

KeinEngelZlat, 2021-10-04:
Thank you, now I will try to install an older version of the JRE. If anything, I will report the result here.

DoctorU, 2021-10-05:
FYI @brumster, 
I'm investigating decompiling the .exe to work out the protocols (having just moved from Windows to Linux) - is there any reason why I shouldn't? (e.g., someone bought the rights to edtracker GUI on winding-up the company)
https://github.com/icsharpcode/ILSpy
I'll let you know if I get anywhere (I have some time this week - fingers crossed!)

brumster, 2021-10-05:
I don't think so; Rob wrote the UI and would be the only person to object, I think. I wasn't involved with the company at the time they closed shop so I don't know what arrangements were made but I seriously doubt anything was sold on to others... ;)

DoctorU, 2021-10-05:
> I don't think so; Rob wrote the UI and would be the only person to object, I think. I wasn't involved with the company at the time they closed shop so I don't know what arrangements were made but I seriously doubt anything was sold on to others... ;)

Magic. I'll see what I can come up with using ILSpy and the [stuff on reddit!](https://www.reddit.com/r/EliteDangerous/comments/jsl0r5/ed_tracker_sketch/)... 😁

brumster, 2021-10-05:
Just mucking around with that utility (ILSpy), being a bit of a java man I can understand what needs changing. In "Form1" there are a number of functions (DownloadStringCompleted) that queried a URL for an XML payload; this XML file I used to maintain on the server, and this would explain what firmwares were available for what devices (6050 based or 9150 based, for example), and where to download the hex image. When you chose to flash a device, it used this XML payload file to download the appropriate firmware that the user had chosen, and flash it into the device.

Basically you can pull any code related to checking and populating that XML payload, and anything related to the flash dialog also (or if someone's really clever, it would be cool to just have the flash feature read a hex file locally). Or for extra credit :) pull the hex files off github here!

DoctorU, 2021-10-05:
I have a murky Java past too... but Python rocks! 😉

brumster, 2021-10-05:
Seriously, if you're in a position to essentially re-build the UI from source, I'm happy to spend a bit of dedicated time working out an integrated method that allows the UI to query that XML payload here on GitHub and use that to drive where it gets the HEX files from (again, the releases folder within this repository) if you are - but would need a bit of commitment from both sides and I think we should be fair to Rob and put the code up here to be maintained going forward...

KeinEngelZlat, 2021-10-05:
@bramster, as far as I understand, I have a problem with the file rxtxSerial.dll ?
[image]

I installed Java version 1.8.0_301-b09, the program worked, but now it gives a problem when trying to firmware in arduino, indicated in the screenshot. I downloaded this file (rxtxSerial.dll) from several places, but the result remained the same. What could be?

brumster, 2021-10-05:
I take it the serial port is correct, so COM3 is your Arduino device? Early screenshots had it as COM16 - is COM3 really the right one?
Maybe try with the "-d" and "-q" options to see if there are any hints as to the problem, but it sounds to me like there is a serial driver issue, the USB driver for the Arduino maybe has an issue. What shows up in device manager when your device is connected?

KeinEngelZlat, 2021-10-05:
Yes, the COM port is correct, just when restarting the PC, the port has changed. The following is displayed in the device manager:
[image]

I also checked the launch of UpdateTool.jar with -d and -q parameters. The -d parameter answered DONE, but the -q parameter gave an error, which can be seen in the screenshot.
[image]

brumster, 2021-10-06:
Sorry, I meant do "-d" (debug) in additional to -q or any other ones you try - might give some more information on what's happening. However, I think this is going to be an issue inside the rxtx driver, or your USB driver/etc looking at that error...

Also might be worth checking your device is actually OK at a simple Arduino level - have you tried flashing anything into it (just a test sketch/etc) from the Arduino IDE, just to confirm that the device is flashing ok?

vanois, 2022-09-01:
> Just mucking around with that utility (ILSpy), being a bit of a java man I can understand what needs changing. In "Form1" there are a number of functions (DownloadStringCompleted) that queried a URL [...]

Hi!
Maybe someone somewhere has this XML file (edflash.xml) saved? It would be possible to restore the firmware function from GUI.

closed 2023-02-16 by brumster