brumster/EDTracker2 #34: STM32F103C6T6

Pheluc, 2021-11-11:
Is it possible to use this code with the STM32F103C6T6?

brumster, 2021-11-11:
Not tried myself but I would say no; it's targeted at Atmel MEGA32's at the moment (the choice was based mainly on their low cost and built-in USB controller). That's not to say it couldn't be refactored to another MCU of course... in theory :)

Pheluc, 2021-11-11:
The Arduino boards had a 100% increase in value in my country, I thought about this possibility precisely because of its low cost and greater capacity. As I'm a beginner I don't understand much of the code.if y get any progress in this attempt, I'll let you know. Wait to arrive  5 units I bought. Thank you for your reply.