brumster/EDTracker2 #37: MPU-9250 sketch is too big for my arduinos memory..

sibang6477, 2022-12-30:
After uploading eeprom_clear sketch which is bundled in example sketches,
i tried to upload mpu-9250 sketch.

However, the error message says that the sketch exceed 102% of the memory.

how can i solve this problem?

(Actually, it does fits when i upload mpu-6050 sketch. But i dont know the difference between existence and nonexistence of magnetometer. Is there big differences?)

brumster, 2022-12-31:
Hiya - this can happen with newer versions of the Arduino IDE that compile a lot more added libs into the binaries. Are you using v1.8.5?

brumster, 2023-02-16:
Closing as no responses; suspect it is down to Arduino IDE compiler as this is a common cause for the problem experienced. Users are advised to stick carefully to the instructions in the readme - specifically, use Arduino IDE 1.8.5

closed 2023-02-16 by brumster