brumster/EDTracker2 #38: I'm so confused, I can't think where to start to adress the problems I encounter, installation problems.

SweetyVolty, 2023-01-10:
Hello, I'm trying to install to a newly bought Arduino Leonardo, it's my 4th arduuino so I'm not a beginner. 

Tried to do as i understood in the readme's or pdfs or even in brumster's videos but not able to write the ino to the arduino at all.. 

Why there is no clear instructions? My most close call was installing IDE 1.8.19 version, created loads of warnings and failed as it not able to find the avr dude folders as it changed some definitions i suppose.. 

I can post the logs btw. 

Then totally bored of ide and tried the cmd updatetool version which resulted another error. 

my line was; java -jar UpdateTool.jar -i EDTrackerII_9250_4.1.0

how am i going to flash this? is there any other way to do this? 

[ide log.txt](https://github.com/brumster/EDTracker2/files/10387114/ide.log.txt)

[hs_err_pid11768.log](https://github.com/brumster/EDTracker2/files/10387115/hs_err_pid11768.log)

SweetyVolty, 2023-01-10:
I'm sorry about my English because my brain is literally dead while trying and trying lol

SweetyVolty, 2023-01-11:
Ok solved this; for the ones confused like me; 

1.8.5 is the solution, delete everything else and all IDE. watch videos for figuring which folder is ending up where.. 

It was momentarily worked; then something happened and head turned upside down, and now it does not want to see the arduino even a little bit. Something went wrong in the calibration i think and corrupted it. Also i cannot get any outputs anymore from ide too.... 

i tried flashing another ino which worked normally and output all the data as supposed to do. i don't know what to try now.... 

except if i can delete the saved configs edtrack4 is not gonna work i think... 

[image]

brumster, 2023-01-11:
As you've discovered, the Aduino IDE versions can performant differently and certainly some of the later versions don't work as the compiled binary is too large to fit on the device. It sounds to me like hardware issues for you - can you link to the specific Pro Micro 16MHz you're using, and the MPU board? How have you built it - in breadboard, just directly wired to one another, or in some sort of PCB? If your comms to the Pro Micro are failing it is probably more related to the cable, the connections or maybe (at a stretch) something driver-related.

SweetyVolty, 2023-01-16:
it turns out that; i was not just a minus 50 IQ idiot; but i was blind too. 

Corrected my incapably bad solders on MCU9250 and than realized i spent more than hours searching for ghost "stuck" settings on cd C: and registry. 

Okay; since I'm ashamed of myself sufficiently there is few points still untouched on my mind; 

I tried hatire arduino with basic ida they provided and than i figured the problem with the EDTracker but weirdly even tho edtracker is more sophisticated than that it still feels incomplete; as I am using an leonardo with MPU 9250 and not only not using the 9-DOF and DMP processing but also it's just feels like 3-DOF but in semi-hibernation. 

i mean; why there is no x,y,z axises? how to add them? etc etc etc.

SweetyVolty, 2023-01-17:
ok 6 hours later; I am extremely positively sure that 9 dofs can be pushed to the extreme steps like this example but i just lost my last two brain cells to kinda understand the system and how it works so i am no nearby compiling something even similar nor making this ready one more beautiful etc. 

 https://x-io.co.uk/

brumster, 2023-01-17:
I think you're falling into the marketing hyperbole trap that is "9DOF is better than 3DOF". There is no 4th dimension that any of these accelerometer/gyro/magnetometer devices are going to be able to magically record or capture for you - all of them are, in the true sense, 3 degrees of freedom devices. Without an outside-in or inside-out point tracker, they are not going to be able to track lateral movements to any sort of accuracy needed for the small and repeated movements of a head tracker for PC gaming. You've got your rotational tracking in 3 angles of rotation, and that's it. Whether it does it with 1 sensor (marketing 3DOF), 2 sensor fusion (6DOF) or 3 sensor fusion (9DOF) makes a difference in terms of the quality and accuracy but at the end of the day, it's still just 3DOF tracking :)

When we designed and build EDTracker Pro, we were aware of all of the devices like x-io on the market and were as confused as anyone else as to why their prices were what they were (always in the hundreds-of-pounds/dollars bracket) for a device that was nowhere near that in terms of production costs. Particularly when most of them were circumventing any sort of CE or FCC approval for selling a wireless device in their market, thereby avoiding an inordinate amount of cost in the development of their device!

Making a cheap, affordable head tracker for PC gaming was/is the only goal of EDTracker - and from the perspective of the open-source DIY one, it had to be exactly that - practical to build and practical to use at the same time. The only way to make them home-build was to use off the shelf components and kits and, put quite simply, to make a *wireless* one that could be put together by a home user was always going to be far too large a unit to be practical on a head (wireless/bluetooth modules, battery module, charge module, compute, MEMS module, plus battery) and also as a result, way more expensive than it needed to be. Hence, EDTracker Pro Wireless, a practical and cost effective way of providing it.

I guess if you're trying to turn the EDTracker project into something else (I'm not entirely sure what you're trying to do with it) then that's fair and totally fine, but it was never intended to be the basis for anything more than the above... ;)

closed 2023-02-16 by brumster