brumster/EDTracker2 #39: Accel values values red in the UI - 6050 EDTracker

kosmonautas112, 2023-02-16:
Hello folks, I'm trying to get my own EDTracker up and going, however I'm running into some issues. With the calibration sketch loaded, the Accel values are in the red regardless of how many times I perform the bias calculations, and the related white dot does not center. Also the hardware sometimes halts after runing for some time, seemingly out of the blue, and doesn't respond to anything until the RESET pin on the Arduino is pulled low or the USB is disconnected and reconnected. As of recent it has been locking up with either both the Arduino RX and TX LEDs off or the RX LED lit solid and TX LED fully out. There have also been times where the Arduino comes up fine (RX and TX LEDs behaving as expected), but there is no data at all coming through. In use is a generic 16MHz Arduino Pro Micro (with J1 to bypass the voltage regulator _not_ shorted (although I doubt this makes any difference)) and an MPU-6050 with everything being connected on a breadboard for the time being, and I have checked my connections several times. Below is an image of the EDTracker UI showing the primary issue.
[image]
Could this be a bad IMU or am I just missing something?

brumster, 2023-02-16:
Hmmm assuming the device is left sitting perfectly still on the table, you should not be seeing any Accel values at all - the accelerometer should be reporting pretty much zero on all axes (maybe a very minor ripple due to movement in the desk/etc, whatever the device is sitting on). But certainly not what you're seeing there. Your tweak values - the numbers to the right of the +/- buttons - are way out though, they should start at zero. You've got 3276 on the X (presumably there's an 8 that's wrapped off the end of that) and a -4915 on the Z. I would tweak those back to zero and start again, or try clicking "restore factory". If they're refusing to stick, it might be that you have some bad eeprom in the Arduino and it can't persist your settings... it might be worth looking into some generic Arduino sketches that validate the non-volatile memory (EEPROM) in the AVR is functioning correctly.

I'd start there. If that all checks out then it *could* be a bosted accelerometer in the MPU but given the gyro/temp readings look fine, I'm not led to think that at first....

kosmonautas112, 2023-02-16:
First of all, thank you heaps for the quick response and the work you do on this repo brumster. I can reassure you that I wasn't sending the breadboard flying across the room to make this issue🙃
As per your pointers I first decided to do a factory restore from the UI, which, as expected, nulled all the values in the boxes to the _left_ of the +/- buttons, however, the Accel values on the _right_ were still up to whatever they are, image below
[image]
Next I decided to close the UI, unplug the board, plug it back in and then open the UI to see if the _left_ side values were saved - and they were showing 0, with the _right_ side values doing as they please, just as with the above image.
After letting the temperature of the 6050 settle, I ran the bias calculation three times, which returned Accel values to the left of the buttons:
[image]
After that I flashed a clean sketch to the board (no code in either of the voids) and then ran an EEPROM test for 1K size, which returned an OK, image attached with a chunk of the code for a sanity check
[image]
After this I adapted the code to write 0s to the entire 1K, which completed successfully.
Thereafter I flashed the EDTracker calibration sketch, opened the UI where the _left_ values were zero but the ones on the _right_ were not, just like in image number one on this comment. Once again, let the temperature settle, run the bias calc 3 times, left values propagated, right side values not showing 0, like in image two. After loads of left mouse clicks to reset the left side Accel values to 0 and settle time, this was the result on the UI:
[image]
At this point I suppose I really should concede and get a new IMU, unless you have something else to try with the current one.
Apologies for the long read and thanks again.

brumster, 2023-02-16:
No, it's no problem, everyone loves a challenge ;)

I take it the IMU is relatively flat, right? Either in the breadboard or sitting on the table, however you've got it (got a picture of the whole setup?).

One thing to possibly try is switching it from "interrupt" mode to polling mode. I vaguely remember seeing some different behaviour for a few users when this was done, which led us to believe some MPUs had issues when reporting via interrupts (my memory is hazy but I do recall there was a reason we put that feature there).

Otherwise I'm thinking to turn on debug mode through the debug serial output, log all the output to a text file for 10 minutes, and then send me the file to load into a spreadsheet and see what the sensor values are doing at a raw level.... just to confirm the MPU is broken/faulty.

As a small aside - why a 6050 instead of a 9250? Just, well... the results are not so great anyway... but if that's just what you had/what's available, that's fine.... but if you're going to resign yourself to getting another one, I'd say stretch the extra pennies to a 9250 if you can... your experience will be much better when you come to use it ;)

kosmonautas112, 2023-02-17:
Test with polling the IMU instead of handling interrupts from it
Temp stab, 3x bias calc, no dice, the Accel is still upset
[image]
Null what I assume are the Accel bias (left hand side) values, power cycle the board, temp stab, no dice
[image]

brumster, 2023-02-20:
> I'd say it's as flat as it gets under normal circumstances - it likely won't be having such a luxury being on my headset. Image of the current setup in all of its glory below!

Oh perfect, thanks. Forgive me asking, but you'd be amazed at what I've seen over the years :D !!

> For the raw data output, which sketch should I load on it? The final use one, the calibration one or does it not matter at all?
Put the final one on, then open up the serial debugger in Arduino IDE and send the character "V" (for verbose) to it. It'll start spitting out a whole load of numbers, as documented [here](https://github.com/brumster/EDTracker2/wiki/Serial-Comms-Protocol), but in short...

Ignore the D lines for now; that's drift compensation but we're far from being there yet
Ignore the T lines, that's the sensor temperature reading
It's the 9 tab-delimited values. The first 3 are the calculated joystick yaw/pitch/roll values (which will be way off I guess), then the 3 raw accelerometer readings and the 3 raw gyro readings.

So I suspect the middle 3 are the ones we're going to be curious about. Log it for a while then dump it into excel and you can graph each of the axis readings over time. If you device is sitting still, then obviously you'd expect very little fluctuation over time. If the accel is all over the shop, then something's just not right.

Send an S (for "Shhh!") when you're done

> With that, you got me thinking, is the 9250 really worth the triple price tag (locally, which is where I'd have to buy to have hope in finishing this while I can, the 6050 is going for 4.05€ and the 9250 is going for 12.83€)? I had thrown the current lot into X-Plane on Wednesday, and even with the bad accel, the tracking performance was surprisingly good, at least as far as I could test it on the breadboard - I didn't notice any jitter at all, probably all thanks to your code. I'll report back with polling instead of interrupts as soon as I'm done testing. Thanks again for all the help.

The main difference between the 6050 and the 9x50 variants if the drift over time - practically eliminated on a well-calibrated magnetometer version, whereas the 6050's we had to use a "drift compensation" feature to estimate the amount the device drifted over time, and then compensate for that itself. Temperature affected it, so you needed to do the drift compensation calculation once it had warmed up. With no external reference point that the magnetometer gains you, there is no way a 6-axis IMU can ever truly know if the gyro has kept it's "true north" accurate or not...

I suspect the DMU processing is filtering out the currently-bad accelerometer values as noise (and rightly so), which is maybe why it's "hanging on in there" even as it stands. The inner workings of the DMU are Invensense-witchcraft that is not made available, but just pushed into the DMU as part of the invensense motion driver (the massive dmp_memory PROGMEM statement found in [inv_mpu_dmp_motion_driver.c](https://github.com/brumster/EDTracker2_ArduinoHardware/blob/master/edtracker/avr/libraries/InvensenseMotionDriver/inv_mpu_dmp_motion_driver.c) )....

Ultimately, if it's working for you right now and you find the drift over time manageable, then run with it. I didn't realise the Invensense MPU6050v9250 prices were so dramatically different these days 😲

kosmonautas112, 2023-03-04:
Apologies for the long delay, life got in the way.
Attached below is the raw data output text file and an Excel spreadsheet with that data imported.
I got a new 6050, but it's showing the same behavior. The data was logged with this new one, as a matter of fact, and I'll keep using it because it sems to be just that bit higher quality.

> Oh perfect, thanks. Forgive me asking,

It's all good, I don't mind you asking at all, I'm happy to provide any information I can to help diagnose this.

>  If you device is sitting still, then obviously you'd expect very little fluctuation over time. If the accel is all over the shop, then something's just not right.

The accel doesn't seem to be fluctuating excessively, it's just reporting high values at all times, even though it's sitting still as a statue.

> The inner workings of the DMU are Invensense-witchcraft that is not made available,

As most things you'd like to have a look at... Open something up and you just find an epoxy blob over the interesting bits. Such is life :D

> I didn't realise the Invensense MPU6050v9250 prices were so dramatically different these days

It's cheaper direct from China, electronics has never really been cheap to buy locally here, but COVID and the chip crisis during/after it likely isn't helping.

I hope you find the logged data useful.

[EDTrackerRawLog.txt](https://github.com/brumster/EDTracker2/files/10889105/EDTrackerRawLog.txt)
[EDTrackerRawLog.xlsx](https://github.com/brumster/EDTracker2/files/10889106/EDTrackerRawLog.xlsx)

kosmonautas112, 2023-04-14:
Hey, I don't want to be rude, but it's been over a month and I'd like to know if anything useful popped up during that period and/or any other steps I should take.

Thanks.

brumster, 2023-04-14:
Hiya, sorry, this dropped off my radar a bit.

I've taken a look at your data. It looks fine :( plotting the X (the joystick output value, the gyro and the accel) they kinda look normal to me. The X creeping off I assume is just because you've not set up any drift compensation yet, so I wouldn't worry about that...

[image]

Y looks fine too :
[image]

...as does Z....
[image]

No wild spikes, and the variance seen there is what I would consider normal. So I don't think your IMU is faulty, or at least certainly not in the sense that it's providing bad data/readings.

If you flash the main sketch in, let it warm up, then do one or two drift compensations, how is the unit behaving, is it still hanging/disconnecting? Just wondering if the problem only occurs in the calibration sketch....

kosmonautas112, 2023-04-15:
All good, thanks for having a look.
And yeah, if memory servers, I just flashed the main sketch and started collecting the data, didn't give it settle time or do the compensation calculation. Good to know that the IMU is working as it should, at least for the most part.
The main sketch certainly seems to be more stable than the calibration one in terms of hanging, however, my main concern currently, as per the name of the thread, is that the accel readings aren't 0 and their boxes are apropriately colored red despite running the bias calc however many times. Also the biases for the accel after the calculation are up in the thousands, and even then the readings aren't 0, with that the accel dot doesn't center. Is there anything that could/should be done about that?

brumster, 2023-04-15:
I'm wondering if it's something in the UI that's the issue, and it's mis-representing the status of the accel and gyro - maybe a bug that's got introduced over the year with a change in the device-side code but without a matching change in the UI side. The problem is I never maintained or had access to the UI code, it was Pocketmoon that did that side of things, and on top of this the 6050 stuff hasn't really been "in focus" just because everyone moved to the newer 9x50 IMUs :(

If the actual function of the device in terms of head tracking (with the main sketch in it) is working, then I wouldn't worry too much about it. I seem to recall Pocketmoon saying once that the MPU fusion that invensense use doesn't rely too heavily on the accelerometer, relying more on the gyro, so the effects of bad accel calibration are probably not too great to cause an issue for most.

Might be stating the obvious, but one accel axis should never be reading (an ideal) zero anyway - whichever axis is perpendicular to 'straight down' is going to always be measuring 1G of downward force, which explains the high value on AZ (although it's still probably offset a fair bit, as per the other axes). We use the 2G full scale range as we never intended EDTracker to be subject to high G forces, so it's in the right ballpark ~16383.

In short, I wouldn't sweat it too much, after 1 or 2 passes in the calibration sketch I wouldn't overly worry, it's probably "close enough" to be accurate for the purposes needed here, and just move to the main sketch and spend some time doing the drift compensation, which is the main thing you need with non-mag devices. It's not perfect, it never can or will be, but it should at least reduce your re-centre keypresses to something manageable in-game (maybe once every 10 minutes or so).

kosmonautas112, 2023-04-15:
I see, fair enough. And yeah, I'm aware that one axis should be sensing the Earth's gravitation.
While digging around myself I found the compiler complaining about some semicolons in the code so I'll probably submit a pull request to get that patched...
One more thing I'd like to clarify before I close this, I should probably keep whichever accel bias it calculates? Don't set it to zero or even touch it and hope for the best, right?

brumster, 2023-04-17:
Typically yes, I'd keep what it's calculated. There are factory calibration values performed and fixed into the MPU at birth, normally any offsets aren't a million miles away but in your case I guess (?) either the factory calibration is way off OR the calculation of the offsets is way off, either due to an issue inside the chip or due to our code. The latter would seem more likely but I'm not aware of this being a systemic problem with the calibration sketch so... yeah...

Ultimately you can try both and see which performed better but I doubt you'll see too much difference ;)

kosmonautas112, 2023-04-17:
Okay, lovely, thank you for your time, thanks for the help.

closed 2023-04-17 by kosmonautas112