brumster/EDTracker2 #40: Axes returning to center

MarceloF1000, 2023-04-16:
All axes always return to center.

I've tried all rotation options. Mine is a MPU6050

I repeated the entire process and still have the issue:
- Erased EEPROM and flash with avrdude
- flashed the calibration hex file with avrdude
- used EDTracker GUI to calibrate. X, Y and Z values on green and bias values on 0. Accel values are all over the place and won't calibrate, but that should not matter.
- flashed the EDTracker hex file for MPU6050
- used EDTracker GUI to run and save drift compensation.
- tested with all rotation options and with auto-recentre off in the GUI

Any help would be appreciated.

MarceloF1000, 2023-04-17:
Actually pitch and roll are the ones returning to zero after movement. Yaw holds position. I have tested this with 2 other sketches. I'm not sure what is going on. Doing research at the moment. If anyone has any clue, please let me know.

Peppie-ZA, 2023-07-25:
I have the same issue, yaw axis keeps centering, pitch works fine.