brumster/EDTracker2 #43: EDtracker Pro Key

Arvigna, 2025-01-02:
Bonjour,
Happy new year !
not sure to be in the good place .... 

here's my problem

I've had an Edtracker for a long time 

I recently took it out again

but as you can see on the image, I can't map a key to stop it.
[image]

Merci.

brumster, 2025-01-02:
I assume the tickbox isn't selectable? You can't check it? Or if you can, do you get any sort of error? You should be prompted to press the key you wish to use. Is that not happening?

Arvigna, 2025-01-02:
Hello!
the tickbox isn't selectable.

I can't check it.

brumster, 2025-01-02:
Ah, ok. I'm not sure I can be much help I'm afraid. Since your problem is with the Pro version, this has nothing to do with me any more, but also I was never party to the UI code so wouldn't know where to start. This might just be down to an issue with software or .NET compatiabilities over the years, I'm not sure. Either way, I'm afraid it's not part of this github repo to address Edtracker Pro problems, sorry :( possibly @pocketmoon might be able to suggest something if he's still around!

Arvigna, 2025-01-02:
ok.
Merci beaucoup.

Arvigna, 2025-01-03:
Bonjour,

my old edtracker pro .

is it compatible with edtracker2? 

Can I update with the files?

starting from scratch ?

thx

Arvigna, 2025-01-04:
> Bonjour,
> 
> my old edtracker pro .
> 
> is it compatible with edtracker2?
> 
> Can I update with the files?
> 
> starting from scratch ?
> 
> thx

Ok.
I have my answer

probably not.

my edtracker pro is not arduino...

am I right?

brumster, 2025-01-04:
Yeah, unfortunately the Pro is completely different hardware - it's ARM-based (Atmel SAMD controller), it's a million miles from the DIY one in terms of firmware I'm afraid! Sorry!

That checkbox not being selectable is not an issue with the EDTracker itself though, it'll be some software issue with the GUI or maybe something that's changed in Windows 10/11 or .NET that's broken it, I suspect. Windows 10 was only just out when we started developing EDTracker Pro 😆 everyone was still on Windows 7/8!

Arvigna, 2025-01-04:
Hello!
yes, I understand.

my edtracker pro 6050 is working fine.

Do you think there is a big difference with version 2?

Would it be beneficial if I gave up my old friend and switched to the new ones? 

Fred

brumster, 2025-01-04:
I presume you mean a DIY EDTracker with the MPU-6050? If by version 2 you mean a DIY EDTracker with the 9x50 MPU, yes, it has vastly reduced drift over time because it uses a magnetometer (the MPU-6050 never had this).

Arvigna, 2025-01-05:
ok.
tank you very much !
Fred